Mair/esp32-course #19: Deep sleep cause restarting  with more than 2100 seconds

Waged, 2020-04-02:
Hi there, I tried this example in your learn ESP32 course and found that when I put a number bigger than 2100 seconds (35 minutes) the ESP32 restarts and keep restarting, could you tell me why ? 

Thanks ,,, 

`void start_sleep_mode(uint16_t sleepInSeconds)
{
  if (sleepInSeconds > 2100)
  {
    sleepInSeconds = 2100;
  }
  unsigned long long mySleeptime = (sleepInSeconds * 1000000);
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_4, 1);
  if (esp_sleep_enable_timer_wakeup(mySleeptime) == ESP_OK)
  {
    printf("Valid sleeping time!\n\n");
    esp_deep_sleep_start();
  }
  else
  {
   printf("Value Exceeds the limit!\n");
  }
}`

Mair, 2020-04-15:
Sorry @Waged I totally missed this.

I'll give it a try and let you know what I find

Mair, 2020-04-15:
your issue is that we are working on a 32 bit system.
try this...
```
uint64_t mySleeptime = 3000 * 1000000;
  printf("time is %llu \n",mySleeptime);
```
the result is not what you expect it to be. Its 18446744072414584320
the reason for this is that each number on the left is considered an int and the result is also an int before its assigned
this will work 
```
  uint64_t mySleeptime = ((uint64_t)3000 * (uint64_t)1000000);
  printf("time is %llu \n",mySleeptime);
```

so your code can change as follows
```c
#include <stdio.h>
#include "esp_sleep.h"

void start_sleep_mode(uint16_t sleepInSeconds)
{
  // if (sleepInSeconds > 2100)
  // {
  //   sleepInSeconds = 2100;
  // }
  unsigned long long mySleeptime = ((uint16_t)sleepInSeconds * (uint16_t)1000000);

  if (esp_sleep_enable_timer_wakeup(mySleeptime) == ESP_OK)
  {
    printf("Valid sleeping time!\n\n");
    esp_deep_sleep_start();
  }
  else
  {
    printf("Value Exceeds the limit!\n");
  }
}

void app_main(void)
{
  printf("Hello world!\n");
  start_sleep_mode(3600);
}

```

closed 2020-04-15 by Mair