Mair/esp32-course #36: Debugging with open OCD.

fahadalisarwar1, 2020-12-29:
Hi this is not really an issue related to the code, but the sequence of chapters. Upto chapter 3, everything seems to be in sequence, however, the chapter 4 "OCD debugging" seems to out of sequence, i.e. its not linking with previous chapters and suddenly the difficulty level in "debugging" chapter rises exponentially.

Mair, 2021-01-18:
Hi Fahad, What do you recomend?

Mair, 2021-01-18:
@fahadalisarwar1 I think I understand what you mean. I realized the first lecture assumes things not yet covered.
I'll get this addressed as soon as I can

fahadalisarwar1, 2021-01-20:
Yes, in the debugging chapter, some aspects are assumed which are not explained earlier. IMHO, the natural place for a debugging chapter is towards the end of course.

Mair, 2021-01-20:
Yeah, I've moved it now. It should make more sense now