Mair/esp32-course #37: Hibernation is not making ESP32 to wake-up again if board is unplugged from power

Waged, 2020-12-29:
Have done the last example of hibernation in the sleep modes and found interesting behavior.

Mair, 2021-01-18:
What dev board are you using. I have found that some dev boards have some strange management around how they manage the battery and actually reset the chip when you plug it in-out.

Waged, 2021-01-18:
I'm using the ESP32-WROVER KIT, the behavior is super annoying as the i wakeup with both timer and a user reset sw on interrupt, if battery died during sleeping mode, the ESP32 cannot wake up until controller reset button pressed and it's not applicable for my use case.

Mair, 2021-01-18:
Yeah that is annoying but, I think its board related. Have you asked anyone on https://esp32.com/ There might be a solution there