Mair/esp32-course #77: No index.html generated for website on chip demo

sbp-dev, 2022-04-17:
Hi, I'm very new to web development. I followed all the instructions to install Svelte, Tailwind and DaisyUI. When I do `npm run dev` , site works fine. When I give `npm run build`, a build folder is generated under site but it has not index.html file so I can't read it from SPIFFS and return to the browser for rendering.
I tried `npm run preview`, site still seems to be working properly on localhost:3000

Attaching screenshot of contents of my site/build directory and assets fetched when hitting localhost:3000
Unable to proceed further, please advise!

[image]
[image]

Mair, 2022-04-17:
hi @sbp-dev please post questions in the discussions https://github.com/Mair/esp32-course/discussions
The index should be there in the root of the build. A few things to check
1. the html files correlate to your svelte files, check you have an index.svelte in your route
2. are you using the @sveltejs/adapter-static
3. delete the build folder and rebuild it. it should be there 
4. failing that clone this repo and see if my code behaves the same way
this is my output

[image]

sbp-dev, 2022-04-17:
Hi Mair, thanks for the prompt response!
> please post questions in the discussions https://github.com/Mair/esp32-course/discussions

Sure, will keep that in mind for any future communication

I tried the following:

- Cloned your repository
- cd into `_15_Internet_Server/v14/_11_web_site_10_change_networks/site`
- `rm -r build/`
- `npm install`
- `npm i @sveltejs/adapter-static`
- `npm run build`
- `ls -alh build/`

Still no .html files! The site works locally though if I give `npm run dev` / `npm run preview`
Also, tried with `npm i @sveltejs/adapter-static --save` and `npm i @sveltejs/adapter-static@next --save` with same results

Please find attached console logs. Thanks in advance.

```
subha@SHADOWBOX MINGW64 /d/Work/embedded/ESP32/projects/esp32-course/_15_Internet_Server/v14/_11_web_site_10_change_networks/site (master)
$ rm -r build/

subha@SHADOWBOX MINGW64 /d/Work/embedded/ESP32/projects/esp32-course/_15_Internet_Server/v14/_11_web_site_10_change_networks/site (master)
$ ls
node_modules/  package.json  package-lock.json  postcss.config.cjs  README.md  src/  static/  svelte.config.js  tailwind.config.cjs  tsconfig.json

subha@SHADOWBOX MINGW64 /d/Work/embedded/ESP32/projects/esp32-course/_15_Internet_Server/v14/_11_web_site_10_change_networks/site (master)
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

subha@SHADOWBOX MINGW64 /d/Work/embedded/ESP32/projects/esp32-course/_15_Internet_Server/v14/_11_web_site_10_change_networks/site (master)
$ npm install
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-freebsd-64@0.14.36 (node_modules\esbuild\node_modules\esbuild-freebsd-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-freebsd-64@0.14.36: wanted {"os":"freebsd","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-android-arm64@0.14.36 (node_modules\esbuild\node_modules\esbuild-android-arm64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-android-arm64@0.14.36: wanted {"os":"android","arch":"arm64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-darwin-64@0.14.36 (node_modules\esbuild\node_modules\esbuild-darwin-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-darwin-64@0.14.36: wanted {"os":"darwin","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-darwin-arm64@0.14.36 (node_modules\esbuild\node_modules\esbuild-darwin-arm64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-darwin-arm64@0.14.36: wanted {"os":"darwin","arch":"arm64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-arm@0.14.36 (node_modules\esbuild\node_modules\esbuild-linux-arm):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-arm@0.14.36: wanted {"os":"linux","arch":"arm"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-ppc64le@0.14.36 (node_modules\esbuild\node_modules\esbuild-linux-ppc64le):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-ppc64le@0.14.36: wanted {"os":"linux","arch":"ppc64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-android-64@0.14.36 (node_modules\esbuild\node_modules\esbuild-android-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-android-64@0.14.36: wanted {"os":"android","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-arm64@0.14.36 (node_modules\esbuild\node_modules\esbuild-linux-arm64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-arm64@0.14.36: wanted {"os":"linux","arch":"arm64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-32@0.14.36 (node_modules\esbuild\node_modules\esbuild-linux-32):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-32@0.14.36: wanted {"os":"linux","arch":"ia32"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-64@0.14.36 (node_modules\esbuild\node_modules\esbuild-linux-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-64@0.14.36: wanted {"os":"linux","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-s390x@0.14.36 (node_modules\esbuild\node_modules\esbuild-linux-s390x):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-s390x@0.14.36: wanted {"os":"linux","arch":"s390x"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-netbsd-64@0.14.36 (node_modules\esbuild\node_modules\esbuild-netbsd-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-netbsd-64@0.14.36: wanted {"os":"netbsd","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-openbsd-64@0.14.36 (node_modules\esbuild\node_modules\esbuild-openbsd-64):
[... cut ...]
> Using @sveltejs/adapter-static
  You should set `config.kit.prerender.default` to `true` if no fallback is specified
  Wrote site to "build"
  ✔ done

subha@SHADOWBOX MINGW64 /d/Work/embedded/ESP32/projects/esp32-course/_15_Internet_Server/v14/_11_web_site_10_change_networks/site (master)
$ ls -alh build/
total 12K
drwxr-xr-x 1 subha 197609    0 Apr 17 20:26 ./
drwxr-xr-x 1 subha 197609    0 Apr 17 20:26 ../
drwxr-xr-x 1 subha 197609    0 Apr 17 20:26 _app/
-rw-r--r-- 1 subha 197609 1.6K Apr 17 18:20 favicon.png
```

Mair, 2022-04-18:
@sbp-dev interesting
have you followed these steps in https://learnesp32.com/videos/15/1_introduction%20to%20svelte%2C%20tailwind%20and%20daisy-ui at 7:40?

Mair, 2022-04-18:
@sbp-dev Ok got it.
This is a recent change made to svelte kit.
In your svelte.config.js under kit add `prerender: { default: true },`

It should look like this
```js
import adapter from "@sveltejs/adapter-static";
import preprocess from "svelte-preprocess";

/** @type {import('@sveltejs/kit').Config} */
const config = {
  // Consult https://github.com/sveltejs/svelte-preprocess
  // for more information about preprocessors
  preprocess: preprocess(),

  kit: {
    adapter: adapter(),
    prerender: { default: true },
  },
};

export default config;
```

Then a re-build and you should be good.

Mair, 2022-04-18:
I'll update the video 
merged #79

closed 2022-04-18 by Mair

sbp-dev, 2022-04-18:
Perfect, this works now!
Thanks again for the quick help.